DccPlusPlus/BaseStation #7: Practicality of using a wireless shield

ryc76, 2016-05-18:
Just wondering how practical it would be to add a version of the controller software that utilises a Wifi shield.

My knowledge of Arduino and C++ is limited so am unsure as to how much work it might involve? My experience is confined to higher level languages.

I suspect the majority of the work would be on the BaseStation side?

Thanks

OzWolfye, 2016-05-18:
I've done a bit of research into doing exactly this.
The sticking point seems to be that both the CC3000 WiFi Shield and the Motor Shield both utilise one  pin in common (from memory, pin 11).
I'm currently looking at using two Arduinos, each having one shield and communicating serially. Unfortunately, I am time poor at the moment so this project has gone on the back burner.
If anyone else can suggest a better way, that would be great.

ryc76, 2016-05-19:
Is the issue with pin 11 fundamental in that it is stopping one stack a WiFi Shield on top of a Motor Shield or is it that the issue is related to requirements of the code?

Thanks

OzWolfye, 2016-06-18:
I believe the problem is that both shields use pin 11 for a different purpose and you cannot assign a different pin to either. Apologies for taking a month to respond.

BatkinRet, 2016-12-10:
(New to Arduino and DCC++)  I would think the solution would lie in ether robust Interrupts, or Multiprocessors. 
Interrupts will raise the issues associated with accurate timing with-in loops, and servicing priority requests.
Multiprocessors mainly raise the cost and inter-processor protocols, but could also reveal Interrupt processing issues.  This approach may be the better solution in the long run as Shields are added to increase capabilities.
  Distracted by getting my first load to work with a Motor Shield with out Load sensing (L298 pins 2&29 not wired)  (Oh the fun of fixing)

EFA57, 2018-07-11:
I thought I saw somewhere that a WiFi should work since it uses the serial pins.

I am hoping to find where I saw that and see if it is for a specific shield. 

Erich
847.341.1120
Also a newbie