DccPlusPlus/BaseStation #8: Missing "#include <EEPROM.h>" in DCCpp_Uno.ino file.

jeffpiep, 2016-06-04:
Compile failed because #include <EEPROM.h> used in Accessories.cpp was not also declared in DCCpp_Uno.ino. 

Once I fixed that, it works just fine. Was able to get a locomotive running quickly with no issues. This is my first DCC experience - your software is great. Thanks!

malc-c, 2016-09-21:
I'm new to using the arduino platform - can you advise how this was resolved as I have the same issue even when I copied the EEPROM.h file from the library folder to the same folder as the base station folder

jeffpiep, 2016-09-21:
With Arduino you need to put all includes into the .ino file. [I learned that here](http://arduino.stackexchange.com/questions/14662/including-libraries-from-header-files-sometimes-doesnt-work). So, just add a line to the top of the DCCpp_Uno.ino file with "#include <EEPROM.h>."  That should fix it. In case it's helpful, I wrote a [short post](http://jeffpiepmeier.blogspot.com/2016/06/ho-trains-and-dcc-part-1.html) on setting the system up - it's not a "how-to," but it does go through the steps I took. Good luck!

malc-c, 2016-09-22:
thanks for the reply... I've uploaded the code now without issue