DccPlusPlus/BaseStation #10: CAB Operation Command REGISTER meaning

Matthysjvv, 2016-08-05:
In the following message:
"<" = Listen to me I am the beginning of a DCC++ command. (A space after < is not required but acceptable)
"t" = (lower case t) This command is for a Decoder installed in a engine or simply a "cab".
"1" = REGISTER: an internal register number, from 1 through MAX_MAIN_REGISTERS (Default is 12), to store the DCC packet used to control this throttle setting (Register "0" is for programming)
"03" = CAB: the short (1-127) or long (128-10293) address of the engine decoder (this has to be already programmed in the decoder) See Programming Commands bellow.
"20" = SPEED: throttle speed from 0-126, or -1 for emergency stop (resets SPEED to 0)
"1" = DIRECTION: 1=forward, 0=reverse. Setting direction when speed=0 or speed=-1 only effects directionality of cab lighting for a stopped train
">" = I am the end of this command

is the REGISTER a different number for every cab you are controlling?

madleech, 2016-08-18:
Yes. My understanding is that each register is used to remember the last command for a loco, so that if the loco loses power momentarily, when it gets power again it will get the speed command again straight away and resume motion. Without this the loco would be very unreliable.

Matthysjvv, 2016-08-18:
So for every loco you add you must use a new register. Loco 1 register 1,
Loco 2 register 2 , etc

madleech, 2016-08-18:
Yes and no. For every loco you want moving _at the same time_, you need a different register. So on a UNO you can have 12 moving at a time. That doesn't count consisted locos; they only take up 1 register since they all respond to the same address. Having 12 independent locos moving at once is getting into fairly serious territory :)

Matthysjvv, 2016-09-06:
Is there any specific decoder that is not working with the DCC++ base
station? I need to convert my n scale loco's to dcc.
Is there any specific decoder that is working well together with the DCC++
base station?