DccPlusPlus/BaseStation #17: Multiple DCC++  cards connected to a PC?

AndrewInEssex, 2017-03-04:
First of all, DCC++ is brilliant! I've looked through the DCC++ commands and I'm trying to see a way to identify the controller; suppose I have two DCC++ base stations connected to a PC, how I can I tell which is which?

-Andrew

gary-menzel, 2017-03-05:
This is probably a naive answer but, as they are connected as a SERIAL device (via the USB port) it is really up to you know which one is connected to which port.  For example, JRMI requires you to select the USB Serial Port of your DCC++ base station.  I'm not an expert in either of these yet - but I suspect JMRI would allow multiple DCC++ base stations to be connected (each having it's own USB Serial Port).  Other than that, as long as you know which one is connected to which port (connect one, note the port, connect the next, etc.) then you just have to send the DCC Command Strings to the right port.

AndrewInEssex, 2017-03-05:
The problem I see is that Windows doesn't always use the same port number even if the same USB port is used; when I boot my machine, I don't know the com port my Arduino card will appear on. Not a problem if there's only one DCC++ plugged in, but with two or more, trial and error seems the only way. 

How about introducing <i> - request board id; returns <d id number>

id number is -1 for a "fresh" board; that's what you'll get reading unwritten EEPROM.

<c new id> - responds <d id number>

The new id number is the id number of the board - 1-254. 

So when a new DCC++ is added, the controller software can do <i> to get the id of the board. If it's been previously identified, then the controller knows which controller it is, if it's id is -1, then the user can be asked for an id number.  The new number is set in the card and next time, it will be recognised - even if it's on a different com port.

AndrewInEssex, 2017-03-06:
I've added this id mod and have a version that supports it. <E> writes the new id and <e> resets the id to -1. So the sequence for connecting a new DCC++ board would be:
PC: < i >
Brd: < d -1 >
PC: < c 20 > (say)
Brd: < d 20 >
PC: < E >
Brd: < e nTurnouts nSensors >

The next time the DCC controller software runs, it looks through all the com ports and sends an < i > command:
PC: < i >
Brd: < d 20 >

The PC now knows the board is the one it used earlier. Assuming the board IDs are stored as part of a layout design, I can copy the design to another PC (eg a spare laptop), plug it into the DCC++ boards and it will recognise them automatically and know which controls which part of the layout.