DccPlusPlus/BaseStation #25: Supported Motor Shields

npayne10, 2017-08-11:
Is there a list of supported motor shields that can be used with this? For instance:
1. 2A MOTOR SHIELD FOR ARDUINO from DFRobot
2. L298 Dual Bridge DC Motor Driver Shield 2A
3. Arduino Motor Shield rev3

chrisridd, 2018-06-13:
The code only supports:

```
//  0 = ARDUINO MOTOR SHIELD          (MAX 18V/2A PER CHANNEL)
//  1 = POLOLU MC33926 MOTOR SHIELD   (MAX 28V/3A PER CHANNEL)
```

The official Arduino motor shield R3 is shown in the videos, so presumably is `0`.

If you know of any boards that behave identically to those two, I'd suggest updating the documentation.

DCC4N, 2019-07-16:
You can find much more motor shields on www.locoduino.org. I am testing a motor shield for nano from Deek Robot or Gravitec.using the 2 channels allow to get 4A.

Nightstalker350119, 2020-02-08:
I saw the need to create a doc, I created a public doc that is editable by anyone here: https://docs.google.com/spreadsheets/d/1aX81e8RtBGNq1Nm77MHVLQLgZCR7iYQC71rElTNzbmw/edit#gid=0

A ticket is open, so everyone can see the list. Unlikely anything will ever get merged.