DccPlusPlus/BaseStation #26: Problem with led near pin 13

Bauche1, 2017-11-06:
Hi there i have problem with my dcc++ i used baseStation 1.2.1..it's recognise loco decoder i can program loco decoder too but my throttle is not working....i tried the D command diagnostic ...it's goes on diagnostic mode ...but i putted an jumper between pin 13 and 5 i tried to send <1> or <0> but my led near of pin 13 is stays on all time...(not going off at anytime) if i put or remover the jumper led is fading a little bit but leds stay all time on...what do you think about?? a defective uno?? thanks for any clue ....regards...

mijnmodelbaan, 2017-11-07:
Do you also have a connection between pins 10 and 12 ?

Bauche1, 2017-11-07:
Yes i have a jumper between 10 and 12...and yes if i send <1> on serial terminal ...all 4 leds goes on...if i put <0> all leds goes off....i tried 2 power supply between 15-18 vdc doing same issue....like 1 said everything fines but in JMRI or DCCpp_controller throttle not working.....question...do i have to cut the 2 brake pad on the back of the motor shield board?

Bauche1, 2017-11-07:
Found my problem...i reset to the default value my decoder....and also... i found i cant use throttle on programming track :) only on main NOW WORKING!!!!!!