DccPlusPlus/BaseStation #28: What voltage should I have coming out of Prog/Main?

Darga89, 2017-11-12:
I am using basestation code 1.2.1 on a Mega with a jumper wire between pins 13 and 2 on a R3 motor shield with the jumper cut all powered with a 15v laptop power supply. I measure 15v going into the board but when turned on through JMRI or sent a 0/1 command through the serial control (all 4 leds light up) I read only about 1.5v on both the programming and main output. That is enough to barely start my N scale Rapido GMD-1 but not my Atlas C630. When I pull the jumper wire I get the 15v output but lose 1 lit led and have no DCC control. I have also tried using an UNO with jumpers on 5/13 and 10/12 but with the exact same issue. Anyone know what the issue might be?

flipperlowe, 2019-02-10:
Darga89 I am having the same problem.
 have you found out how to correct this issue??

johncolman, 2019-02-10:
Have you got two power supplies. One for the UNO 5v and the 12-15v DC on the motor shield. I think the original programmer is no longer on the scene so try www.Trainboard.com as there are programmers there that have taken this a step further.

flipperlowe, 2019-02-11:
John

Thank you for directing me to the train board; a lot of info on there and a people who are willing to help and not make a a person feel dumb