DccPlusPlus/BaseStation #34: I would like to know if i can add a WiFi shield so i can access from my phone/tablet

EFA57, 2018-06-21:
I would like to know if i can add a WiFi shield so i can access from my phone/tablet.

I want to try using controller ide using apde app.

Thanks,

Erich

DCC4N, 2018-11-19:
I found this project which use an ESP8266 as an access point to connect the WiThrottle iPhone app 
[https://github.com/vhar/withrottle](url)

Do you know this project and other projects like this one ?

Thanks,
Dom

atanisoft, 2018-11-19:
> Do you know this project and other projects like this one ?
https://github.com/atanisoft/BaseStation

This integrates the ESP8266 with the base station but doesn't have WiThrottle support (you can connect JMRI to the base station wirelessly and use it's built in WiThrottle server though)

DCC4N, 2018-11-19:
Thanks Atanisoft,
I am looking for the support of WiThrottle (iOs) and EngineDriver (Android) directly to DCC++ without JMRI.
I will see also you DCCppESP32 project ([https://github.com/atanisoft/DCCppESP32](url)) but I would like first to learn with 3 basic circuits : ESP8266 (ESP-01 or 02) supporting the webserver and converter between the WiThrottle and DCC++ interfaces , Nano for supporting DCC++ and L298 power for the tracks.
Dom

atanisoft, 2018-11-19:
> I am looking for the support of WiThrottle (iOs) and EngineDriver (Android) directly to DCC++ without JMRI.

https://github.com/msunderwd/BaseStation-Uno/tree/WiThrottle

That may work but it doesn't have ESP support so you would need to sort that part out.

For the Nano you can look at https://github.com/skyne/BaseStation

EFA57, 2018-11-19:
Is there a way to use JMRI on an android? 

I am looking on the JMRI web site now

DCC4N, 2018-11-19:
> > I am looking for the support of WiThrottle (iOs) and EngineDriver (Android) directly to DCC++ without JMRI.
> 
> https://github.com/msunderwd/BaseStation-Uno/tree/WiThrottle
> 
> That may work but it doesn't have ESP support so you would need to sort that part out.

[https://github.com/vhar/withrottle](url)
This one seems to do what I expect. Do you know it ?

Thanks
Dom