DccPlusPlus/BaseStation #36: Arduino UNO + L298P R3 Motor shield : USB Serial Comms fail

ProfKlyzlr, 2018-10-09:
Dear DCC++ team,

First foray into DCC++,  and kinda feeling like I'm hitting most of the trees on the way down.
The system:
- Arduino UNO
- Arduino L298P R3 motor shield (clone)
- JMRI 4.12 and 4.13 tested on both Win10 Home x64 and Raspberry Pi (mSteveTodd RPi image)

The core issue at the moment seems that after only a few minutes, the serial connection from the DCC++ components to the JMRI Host has some form of hiccup, and is unresponsive until JMRI is restarted.

Typical signs that it's broken are:
- Loco on Main track no longer responds to commands from JMRI throttle
(continues at last speed/direction value)
- JMRI "Track Power" indicator changes to Yellow ("Unknown") state, and will not return to either RED ("Off") or Green ("On") state.

I have tested :
- multiple USB cables, 
- a pair of different Arduino UNOs, 
- both PC (Win10 Home) and Raspberry Pi JMRI Hosts
- both having and not-having a loco on the Main Track output

and the symtoms stay constant, only with slightly-varying times between JMRI-launch and comms drop-out. 

The "VIn" contact-point has been cut, allowing the Motor Shield to be powered by a  15VDC Regulated 2A switchmode plugpack, while the Arduino UNO is powered via the Host's USB connection. The "Brake" contact-points have not been cut, but there is nothing connected to those pins (Literally just the Arduino UNO and the Motor Shield). The topside Pin 5<>13 and 10<>12 jumpers are in-place.

Thru consultation with mSteveTodd (Mr "EngineDriver" app dev, Many Thanks!), the logs from the various JMRI instances and tests suggest a failure of the Serial Comms, possibly in the "reporting back Arduino > JMRI" half (while it's working, JMRI/ED throttle commands are received, actioned, and the loco moves as-expected with a reasonable reaction-time to changes in throttle setting).

In all instances, the JMRI "DCC++" settings show a baud rate of 115200 in a greyed-out field,
so I take it that this is both correct, and cannot be changed from the JMRI Preferences dialog.

Attached are some ZIP files of logs from both the RPi and Win10 test rigs, for reference.

Questions:
- What are the known symtoms of using a "bad" (cheap, clone) L298P R3 motor shield with a UNO in a DCC++ rig? Is unstable/unreliable USB Serial Comm performance one of them?
(If Yes, is there a Verified list of absolutely Known-Good working Motor Shield options and sources?)

- Is it possible that a oddball PSU connected to the Motor Shield could cause the Arduino USB Serial Comms to mis-behave? I have tried adding filter caps to the DC-Input of the Motor-Shield in a "blind guess" try to smooth out any switchmode PSU ripple, but no change.

- Could a bad/incorrect USB Arduino UNO driver be to blame? It feels unlikely as I'm seeing the same issue with the same DCC++ hardware connected to both a RPi on Linux and a quadcore laptop running Win10 Home x64, but still possible?

- Any other suggestions, thoughts, or diagnostic tests I should try?

Many Thanks and Happy Modelling,
Prof Klyzlr

[ProfK_Win10-Home_x64_DCC++_test-logs.zip](https://github.com/DccPlusPlus/BaseStation/files/2460380/ProfK_Win10-Home_x64_DCC%2B%2B_test-logs.zip)
[ProfK_RPi_DCC++_lsusb_and_Full-Debug_logs.zip](https://github.com/DccPlusPlus/BaseStation/files/2460381/ProfK_RPi_DCC%2B%2B_lsusb_and_Full-Debug_logs.zip)
[ProfK_RPi_DCC++_quick-fail_log-files.zip](https://github.com/DccPlusPlus/BaseStation/files/2460382/ProfK_RPi_DCC%2B%2B_quick-fail_log-files.zip)

atanisoft, 2018-10-09:
this very likely should be routed to the JMRI team instead of DCC++. It would appear that JMRI is losing its serial connection to the base station but the base station is still behaving normally.

As for your questions, I have not seen any issues with the inexpensive clone boards (Uno, Mega or motor shields). If the Arduino is powered by the USB that shouldn't be a problem for test envs (it is how I power mine as well as the ESP32 boards I test with) but long term setup should be through a dedicated power supply.  For the possibility of a driver issue, if you use another program outside of JMRI that can access the serial port (like MTPuTTY or the Arduino IDE) you can confirm that the basic operation of the base station manually outside of JMRI.

scanlonjrm, 2019-07-23:
I have checked the arduino uno and it matches the one prof klyzr says should work BUT it doesn't
I'm new to using arduino, I have build the DCC++ base station and tested it. I have installed the JMRI software and it connected to a laptop running windows 10.
I keep getting these errors from DecoderPro
Unable to create connection "DCC++" (D)
Serial Port COM# is in use
System connection DCC++ provides a null manager for interface jmri.commandstation
Unable to run startup actions due to earlier failures
I have installed JMRI on 2 different laptops with the exact same error messages
I've checked device manager and it looks ok nothing else is using the port.
I'm very frustrated having spent several hours trying to resolve this...
Any thoughts out there?
Thanks Jim