DccPlusPlus/BaseStation #37: Decoder reading and writing CVs

WhitePassRanger, 2018-10-13:
I've seen a few videos online that DCC++ can read and write cv's, but I cannot get it to work.  I am currently using the arduino and a command station, it will run the trains great, even via my phone!  Is there something I need to do differently to read and write? I get "123" on every field I try and read.

Not sure if this is my operation issue or if its still under development.

Thanks for your help in advance!

atanisoft, 2018-10-13:
That looks correct from what I can tell. Check the serial console output for any clues.

WhitePassRanger, 2018-10-13:
I get output from Arduino serial monitor but can't run JMRI because the port is busy.  Is there another way to run JMRI and monitor at the same time?

Sent from my iPhone!

"Ruthlessly compete with your own best self!" Apollo X Engineers

WhitePassRanger, 2018-10-13:
Okay, now I'm feeling really green!!  How do you close the Arduino controller without the serial monitor shutting down?

EFA57, 2018-11-19:
Sorry, 

I just saw your update.

As long as the Base Station is running and connected to the PC you should be able to run JMRI ie DecoderPro and set it to the same port as the IDE & the Sketch and let DecoderPro know to access DCC++ and follow their instructions to access programming track and then read the decoder.

You MAY need to up the current from the default 300 to about 750 for HO.

Also make sure you are powering the Motor Shield with around 12 v min.

WhitePassRanger, 2018-11-19:
No dice, JMRI will not access the port because it says it is already in use and opens into preferences to select a different port.