DccPlusPlus/BaseStation #41: Mega+ESP8266 Board

EFA57, 2018-12-26:


EFA57, 2018-12-26:
I purchased a Mega+ESP8266 all in one board and downloaded the basestation and added my motor shield and it worked.

I have no idea about the ESP8266.
I think from what I read I think it gets programmed using python but since I am still not fluent in C++ I am unsure of the next steps in programming and then seeing if I can access using my Engine Driver on my phone/tablet.

Is there somewhere where I can get some interactive help?

I would love to even set a time to text back and forth when I can focus on my computer (when my family is not around).

Thanks,

Erich
847-341-1120

austwhite, 2021-12-09:
I know this is 3 years old, but check DCC++EX if anyone else sees this issue,  It is based on this project and has the option to use the Mega WiFi (Mega with 8266).
I use it and it does work :)