DccPlusPlus/BaseStation #45: Not reading CVs with some DCC decoders

olc, 2019-05-05:
Hi!

Not sure it is an issue with DCC++ or something else.

Here is my setup: 
- Arduino Uno
- Deek Robot motor shield (modified according to the documentation instructions)
- 3 minitrix engines (N scale) with mTc14 DCC decoders

The 3 engines can be controlled without problem on the main track. But I am not able to read CVs nor engine address from two of the engines on the programming track. With these two engines, motors are even not activated which seems to prove that DCC decoders are not able to decode the DCC command.

With the working engine, sending `<R 1 666 777>` gets back `<r666|777|1 3>` which is correct.
With the not working engines, sending the same gets back a failed `<r666|777|1 -1>`.

I have inverted DCC boards in order tocheck if problem could be at the engine side: the DCC board which is readable can always be read regardless of the engine. 

I have no problem reading the CVs from any engines using the original Mobile Station Throttle.

Weird, isn't it? Any help would be greatly appreciated. Thanks in anticipation!

-- 
Olivier

tomcevans, 2019-07-15:
Have you been able to read any CVs using the DCC++? I looked up the Deek Robot Motor Shield, and I don't see any hardware in that schematic for sensing current to the motors. DCC++ requires a motor shield that can sense the motor current in order to read engine responses to DCC programming.

positron96, 2019-07-29:
He might be talking about htis shield: https://www.youtube.com/watch?v=rWnYqrK3prk it seems to be a copy of Arduino Motor Shield V3

g0rt, 2019-07-30:
I guess he'll need to specify which one he's using, since the diagnosis depends on the model.