DccPlusPlus/BaseStation #46: Arduino DCC++ base station Serial Port in use

scanlonjrm, 2019-07-23:
I'm new to using arduino, I have build the DCC++ base station and tested it. I have installed the JMRI software and it connected to a laptop running windows 10.
I keep getting these errors from DecoderPro
Unable to create connection "DCC++" (D)
Serial Port COM# is in use
System connection DCC++ provides a null manager for interface jmri.commandstation
Unable to run startup actions due to earlier failures
I have installed JMRI on 2 different laptops with the exact same error messages
I've checked device manager and it looks ok nothing else is using the port.
I'm very frustrated having spent several hours trying to resolve this...
Any thoughts out there?
Thanks Jim

g0rt, 2019-07-24:
the first thing that comes to mind is that if Windows says the serial port is open, then some other poece of software has the serial port open. Are you using a genuine Arduino with the Arduino software? If so, can you send commands to the DCC++ using the Arduino serial console? If so, did you close the Arduino serial console before starting the JMRI software, since they both use the same serial port?

PeterSek, 2019-07-24:
Please try this
unplug all usb/serial connections
and only leave the DCC++ usb cable in place (plug usb before starting JMRI)
perhaps reboot before this step
then start JMRI and check to make sure it points to correct serial/usb
PanlePro->Edit->Preferences->Connections

Peter

scanlonjrm, 2019-07-24:
g0rt, thx I thought of that can’t find anything holding the port software or hardware. yes genuine arduino/arduino shield, using serial console to communicate and yes closed console and arduino software too.
PeterSek, thx I tried that to everything but DCC++ shutdown, pulled power cord and battery held down power button 30 secs restored everything and restarted. Started JMRI checked was connected to com3 still got same errors. 
Thx, I appreciate the suggestions any other thoughts? I’m going to try a third laptop later today to see if I get different results.

g0rt, 2019-07-24:
You can isolate the problem by attempting to connect to the Arduino board with a terminal program, such as Tera Term Pro. You can download TTP v3.1.3 for free from ayera.com. After shutting down the Arduino development environment and serial monitor, try connecting to your COM port with Tera Term using the same serial port settings that your serial monitor is configured for. If Tera Term can open the COM port (and you get the standard response from DCC++) but JMRI cannot, then it is an issue between JMRI and the COM port. That could mean either the JMRI code or the Java Runtime Environment you're using. In my setup, I have DecoderPro v4.14 using Java v1.8.0_202 (64-bit). You said you are running Windows 10, but can you provide more info on the version (i.e. 32 or 64-bit), the version of Java you're using, and the version of DecoderPro you're using? You can find most of this out by running the JMRI InstallTest program.

All that being said, if I run the InstallTest program with DecoderPro already running and connected, I get the same message you are getting, but that is because DecoderPro is running.

scanlonjrm, 2019-07-24:
Using a terminal program is a great idea I will give that a try. The system is a toshiba satellite 64 bit 8 GB ram with a 500gb solid state drive. I have installed DecoderPro v4.14 and java v1.8 (verified at download) like you have, in fact just to be sure I uninstalled both and reinstalled them again., after the firs failure. Then I checked and installed the most recent win 10 updates too. Still have the same issue. I haven’t run JMRI Installtest program yet. I’ll try the terminal program, a different slightly older 32bit windows 10 laptop and maybe a raspberry pi zero over the weekend and let u know how it turns out. Thx

scanlonjrm, 2019-07-28:
Ok, so here is what happened, after several hours of toying around and trying with the second laptop worked it kinda worked. I got the DecoderPro to talk to the arduino base station but could only communicate on one of the ports, very wierd. If I switched back and forth between the 2 ports I could get a loco programmed on say port "A" using a short piece of track, so some progress. However nothing would respond on port "B". Long story short I figured out something on the motor shield was touching and shorting against the arduino uno. When I separated them (pulled the motor shield out from the arduino about 1/16 of an inch) everything started working. I connected to the programming track and the main track and was able to program locos on the program track and run them on the working track as well. Having gotten it to work I wanted to see what was up with the original laptop so I took the original laptop that wouldn't communicate and reconnected it to the uno and DecoderPro worked perfectly. 
So it appears that if you press the arduino motor shield into the arduino uno to hard something shorts and the intermittent issues occur. It's also interesting that the motor shield has a plastic guard on it which I assume would isolate it from the uno but apparently not enough.
Thanks to all who worked with me to address this and I hope this helps someone else as well.