DccPlusPlus/BaseStation #49: Return -1

Franky1265, 2020-02-06:
Hi, first thanks for that cool project.
I bought an ardolino uno, a ardolino standard shield, and use jmri.
Within 2 days all woks fine except the programming Read cv
I have 16 volt dc, and i can program any cv. But when i want to read at the programmtrack. The motor ist moving sime milimeter. But i get back -1 no reply. It is mo contact problem as i connected the lok with krokodiles. I used the master ardulino software.
Can anybody give me a hint

Franky1265, 2020-02-07:
Hi, i went on but did not solve the issue.
When i send <R 1 123 123> In the ardolino stetch i also get back -1
And the power at a1 does not change dc or ac. Voltage. The lok is jumping.
I opened the 2 brak bridges and the voltage bridge.
Can now anybody help me please
Regards Frank
[image]
[image]

Kcsmith0708, 2020-02-08:
Franky, 

I haven't been following your current project but I built several Uno and Mega DCC++ systems. 

I've never cut the break point on the back of any of my DC motor shields. Does the project your working on require you to do this.? 
If not, try a simple test and reconnect the break by either re-soldering or tape wire across the point to close the connection. 
Leave the VIN point cut or open. 

Reassemble and test the code as is. 

Regards, 
Kevin

Franky1265, 2020-02-09:
Hi Kevin,
My project is just running dcc at HO
I connected the 2 break point again,
Still the same, all works except the reading cv
I tried Lenz silver esu lp 2 and 4 at least the lenz lok is jumping

Franky1265, 2020-02-09:
Does anybody know how the voltage at A1 is moving when i resd cv.
Is it measurable wit a normal voltmeter, is it dc or ac,

Franky1265, 2020-02-10:
So, i looked a0 there dc is increasing till 1,4 volt . When droving. At a1 i see no milivolt even no flipping with the osziloskop. The parrot is dead. I come back when i changed it

Franky1265, 2020-02-12:
Me again, the motorshield was defect. Now it works. And a1 is flippind with a 5 euro multimeter to 0,2 volt dc when i read cvs

Kcsmith0708, 2020-02-14:
Franky, 
Congratulations, 
Good Job finding the problem. 
Enjoy DCC++ 
Kevin