DccPlusPlus/BaseStation #56: Large DCC Packets are not transmitted correctly

HFF-Git, 2020-08-16:
There are DCC packets of up to ten bytes of data. For example the CV XPOM command. The original code does not handle
such large packets. I have adapted the "loadDccPacket" routine to generate the bit stream. If there is interest to include it, just let me know.

DCC++ is just great for starting digital model railroading!!!!

atanisoft, 2020-08-16:
@HFF-Git Do you have some references on where that packet format is defined? The NMRA specs declare the maximum data portion of the packet is six bytes IIRC.

HFF-Git, 2020-08-16:
Please see: RCN214, RailCommunity. https://www.railcommunity.org. They follow the NMRA specifications but seem to have many ( newer ? ) instructions for advanced decoders...

The command is:

1110-KKSS VVVV-VVVV VVVV-VVVV VVVV-VVVV {DDDD-DDDD {DDDD-DDDD {DDDD-DDDD {DDDD-DDDD}}}} 

( first byte -> instruction byte ) It allows the quicker access to CVs that you only reach via CV #31 / 32. This command is a maximum of 8 bytes long, adding  1 or 2 two bytes for the loco address -> 10bytes.

I stumbled across this issue when using the command to send speed, direction and functions F0..F12 in one command. The NMRA states that the function F0 .. F12 should be refreshed. 

This command does it. And it has - worst case - 8 bytes. 

0011-1100 RGGG-GGGG DDDD-DDDD {DDDD-DDDD {DDDD-DDDD {DDDD-DDDD}}} 

( first byte -> instruction byte )

The NNMRA specs also defines only up to 28 functions. The RCN spec lists up to 68 functions. 

Hope this helps.

atanisoft, 2020-08-16:
> The NNMRA specs also defines only up to 28 functions. The RCN spec lists up to 68 functions.

That isn't entirely true, the NMRA spec declares functions 0-28 but does not limit to only those 29 functions. It is defined as up to 64k functions but very few decoders use beyond the declared 0-28.

> Please see: RCN214, RailCommunity. https://www.railcommunity.org. They follow the NMRA specifications but seem to have many ( newer ? ) instructions for advanced decoders...

XPOM looks like it would fall into the "future expansion" category on the NMRA specs and would appear to be in direct violation of the NMRA standards (limits to three byte payload). As such, it is very likely not going to be supported by this code base (especially considering it is more or less abandoned!).

Do you have a specific decoder that requires the usage of XPOM rather than using NMRA standard CV access commands?

I'd suggest file an issue over on https://github.com/DCC-EX/CommandStation-EX since that is being maintained.

HFF-Git, 2020-08-16:
I have not gotten around to play yet. Nevertheless, the RailCommunity is the association of German model railroad equipment manufacturers. I have seen ads where they offer 68 functions. I'd rather implement the extensions and be done with it. Thanks for the tip to file the issue in the other place.

atanisoft, 2020-08-16:
@HFF-Git Correction, use https://github.com/DCC-EX/BaseStation-EX instead of https://github.com/DCC-EX/CommandStation-EX.

HFF-Git, 2020-09-05:
This issue was moved.

Correction, use https://github.com/DCC-EX/BaseStation-EX instead of https://github.com/DCC-EX/CommandStation-EX.

closed 2020-09-05 by HFF-Git