DccPlusPlus/BaseStation #58: No software license included in repository

saxbophone, 2020-09-25:
Including a software license in a project gives other developers who could make use of it more confidence about under what terms they may use it.

If you create the file `LICENSE` within the Github UI, Github will give you access to a bunch of templates of popular open-source licenses that you might want to use, with summaries of each.

Might this be something you could do?

FrightRisk, 2021-03-05:
Project was abandoned in 2015-2-16. See https://dcc-ex.com instead

closed 2021-03-05 by saxbophone

positron96, 2024-08-10:
I've been wondering about this project's license for a couple of years, until I've recently found its now dead googlesites project page. It's archived here: https://web.archive.org/web/20200919034010/https://sites.google.com/site/dccppsite/

It says this:

> This software is freely-available from the [DCC++ GitHub ](https://web.archive.org/web/20200919034010/https://github.com/DccPlusPlus)site and is provided under standard GNU open-source licensing: use it, edit it, hack it, redistribute it. If you want, you can even build it into commercial products.

I think it's poor wording to say that the license is GPL. But since the site doesn't call the license by name, you can take the text literally as "use it, edit it, hack it, redistribute it,.. build it in commercial products".

saxbophone, 2024-08-10:
Even if they were referring to the GPL, that doesn't preclude commercial
usage. The GPL allows commercial usage of the software and software that is
based on it. But derivatives and software that links to them must also be
GPL which can discourage (but not *prohibit*!) commercial use.

But this is all moot hypothetical really as they don't name a specific
license. Even "GPL" isn't clear enough as GPLv2 and GPLv3 are in wide use
and the difference between the two is not insignificant in terms of its
implications for licensing and reüse of the work.