DccPlusPlus/BaseStation #59: Centipede Shield V2 - support

ear9mrn, 2021-01-30:
I have modified your great code ever so slightly to provide support for the Centipede Shield V2. With the uno this enables expansion from 20 digital IO pins to a total of 84 :-) 

I was wondering if you would allow a) take a look at how I have achieved this (I am a no coding expert) and b) if this should be added to the base code or if I can create a fork?

Thanks,

Pete.

FrightRisk, 2021-03-05:
This software is long abandoned. The project development was picked up by DCC++ EX. You can find us there. https://github.com/DCC-EX/CommandStation-EX https:://dcc-ex.com

austwhite, 2021-12-09:
Agree with @FrightRisk    
DCC++EX is very actively developed, and they have a bug-fixed version of this project if you prefer this over the EX version.
Personally, I upgraded to DCC++EX and have never looked back.  It is so much faster sending codes and commands to the loco's, especially if you have more than 2 or 3 loco's running